OctoPrint/OctoPrint-MQTT #140: Naming warnings for MQTT when using with Home Assistant

CobraA1, 2023-08-05:
So I'm running this plugin to provide MQTT for Home Assistant, and it works great - but I'm now getting a warning about changes coming in the future. Can this be investigated?

More information about the naming changes can be found [at their website](https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/).

```
Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

    button.octoprint_mqtt_reboot_system
    switch.octoprint_mqtt_pause_print
    sensor.octoprint_mqtt_print_status
    sensor.octoprint_mqtt_approximate_completion_time
    button.octoprint_mqtt_shutdown_system
    switch.octoprint_mqtt_connect_to_printer
    sensor.octoprint_mqtt_tool_0_target
    switch.octoprint_mqtt_camera_snapshot
    sensor.octoprint_mqtt_last_event
    sensor.octoprint_mqtt_approximate_total_print_time
    sensor.octoprint_mqtt_current_z
    sensor.octoprint_mqtt_tool_0_temperature
    sensor.octoprint_mqtt_print_file
    sensor.octoprint_mqtt_bed_target
    button.octoprint_mqtt_restart_server
    button.octoprint_mqtt_cancel_print
    binary_sensor.octoprint_mqtt_connected
    sensor.octoprint_mqtt_slicing_file
    sensor.octoprint_mqtt_print_time
    sensor.octoprint_mqtt_slicing_progress
    sensor.octoprint_mqtt_print_time_left
    button.octoprint_mqtt_emergency_stop
    sensor.octoprint_mqtt_bed_temperature
    sensor.octoprint_mqtt_print_progress
    sensor.octoprint_mqtt_soc_temperature
    binary_sensor.octoprint_mqtt_printing_2

```

jneilliii, 2023-08-05:
this has nothing to do with the MQTT plugin does it and more to do with the [Octoprint-HomeAssistant](https://github.com/cmroche/OctoPrint-HomeAssistant) plugin? This plugin literally just sends data out, it doesn't register with HA or anything like that.

jneilliii, 2023-08-05:
there's also already a PR submitted to that plugin as well that was going to be reviewed this weekend I've been told.

https://github.com/cmroche/OctoPrint-HomeAssistant/pull/113

closed 2023-08-05 by jneilliii