OctoPrint/OctoPrint-MQTT #142: Since you haven;t updated your plugin in over a year, are you going to update it for the breaking changes in Home Assiatant?

nerlins, 2023-08-18:
See this post over in HA:

https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099

Your plugin is causing this error message in Home Assistant:

> Discovered entities with a name that starts with the device name
> 
> This stops working in version 2024.2.0. Please address before upgrading.
> 
> Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

jneilliii, 2023-08-18:
This plugin is not the home assistant plugin. That one can be found at https://github.com/cmroche/OctoPrint-HomeAssistant.

nerlins, 2023-08-18:
Sounds good. Redirecting all issues to the HA plugin.

closed 2023-08-18 by nerlins