OctoPrint/OctoPrint-MQTT #146: Entities Unavailable in Home Assistant

broyuken, 2024-01-09:
When my printer is not printing, many of the entities in Home Assistant go unavailable. The Cancel, Pause, and anything related to print time are unavailable. I have the retain setting on, could we change the settings so they default to a value of 0 or something instead of going unavailable?

[image]

RoboMagus, 2024-01-17:
Sounds like this is working as it should to me. If there's nothing to pause or cancel, the ability to do so is unavailable and thus reflected as such.

A more logical way to get rid of unavailable entities in your HA dashboard would be to hide the entities if nothing is printing (using e.g. the `conditional` card).

broyuken, 2024-01-17:
They’re not in my dashboard, but I have a page that shows all unavailable entities. I may have to just filter these out, but I was hoping they could have a default value of 0 or something instead of just going unavailable.