OctoPrint/OctoPrint-MQTT #147: Since the last update I see constatly this in logs and don't get any updates via MQTT

guinnberg, 2024-01-29:
#### What were you doing?

Just updated to latest version and I don't get MQTT updates anymore

#### What did you expect to happen?

Get MQTT messages

#### What happened instead?

I can see this in the octoprint.log

`2024-01-26 18:47:34,251 - octoprint.plugin - ERROR - Error while calling plugin mqtt
Traceback (most recent call last):
  File "/home/ivan/oprint/lib/python3.9/site-packages/octoprint/plugin/__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/ivan/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/ivan/oprint/lib/python3.9/site-packages/octoprint_mqtt/__init__.py", line 174, in on_event
    self.progress_timer = RepeatedTimer(5, self._update_progress, [payload["origin"], payload["path"]])
TypeError: 'NoneType' object is not subscriptable`

#### Version of OctoPrint

1.9.3

#### Version of the MQTT plugin

0.8.14

#### Used MQTT broker and its version

Mosquitto 6.4.0

jneilliii, 2024-01-29:
I think this may be a result of the event FILE_DESELECTED in the conditional [here](https://github.com/OctoPrint/OctoPrint-MQTT/blob/d31d612d72b477d109e8dc6fcfe7aa578ca47d75/octoprint_mqtt/__init__.py#L172). Looking at the docs it seems there is possibly no payload in that event. Will make a quick adjustment and get you to test a separate branch install.

jneilliii, 2024-01-29:
Try installing the version here and let us know if the error is still showing once updated. From Plugin Manager click get more button and at the bottom in the ...from URL field copy/paste this link and click Install. 

```
https://codeload.github.com/OctoPrint/OctoPrint-MQTT/zip/refs/heads/bugfix/FileDeselected
```

guinnberg, 2024-01-30:
I don't see any errors anymore, but I still don't get new MQTT messages and everything shows as unavailable in Home Assistant.

MQTT explorer for reference:
[image]

guinnberg, 2024-01-30:
I've done a clean install of the latest release and I can see this in Octoprint when I enable DEBUG logging:
> 2024-01-30 11:53:33,126 - octoprint.plugins.mqtt - DEBUG - Not connected, enqueuing message: octoPrint/temperature/tool0 - {"actual": 22.62, "target": 0.0, "_timestamp": 1706615613}

I can't see anything on my Mosquitto logs... I've also tried to provide wrong credentials and still can't see any extra errors either in Octoprint or Mosquitto

AWSW-de, 2024-06-07:
Hello, 
I see the same issue here.

My question is, if this is a still maintained plugin, because this issue is open since January…. ?

Kind regards and thanks for a solution in advance.

jneilliii, 2024-06-07:
need more details related to the connection probably. last I tested the plugin everything was working fine, so need to get full octoprint.log from anyone experiencing this issue with debug logging enabled for the plugin in OctoPrint's logging section of settings.

jneilliii, 2024-06-07:
Confirmed, I just did a fresh install of MQTT on a device that has never had it installed before and it's working fine for me, so definitely will need more logs. Keep in mind, the latest stable version 0.8.15 will still have the initial error posted in this thread with file deselect. 

[image]